rdkcentral/rdk-halif-test-device_settings #49: Updating the L2 Lowlevel Spec and L2 Test code after ChatGPT generation.

srinivasgtl, 2024-06-06:
This task is to update and enhance the L2 low-level design and L2 Test code generated by ChatGPT.

1. This task will add manual effort to fix the missing documentation issues.
2. Any issues reported during testing.
3. Upgrading the flow diagrams wherever required.

mhughesacn, 2024-06-06:
Hi Srinivas: I recently asked Comcast Legal about the use of AI-generated code (on foot of a conversation with GeraldW) and their response was: " So, for now, our stance should be that AI-generated code should NOT be submitted to RDK repos."  We (CMF) have requested some more guidance from Legal and expect to update our documents very soon.
- Martin Hughes, RDK CMF Compliance Team

Ulrond, 2024-07-01:
@srinivasgtl to review

srinivasgtl, 2024-08-16:
This issue is moved to different PRs and branches creating one PR per module.  There is no need for this issue further.

closed 2024-08-16 by srinivasgtl