rdkcentral/rdk-halif-test-device_settings #391: Bug: <Short summary of the problem>

anburaj17, 2025-04-04:
### Problem/Opportunity

VTS testcase have hardcoded the enhancement range value 16 in testcase code. It should be retrieved and validated from profile file since each platform might have a different capability.
Realtek SOC will support the enhancement range till 12 only. So, the remaining 4 iteration checks will fail.

**VTS versions used:** 3.1.2

### Steps to reproduce

Run L2 ‘set dialog enhancement’ VTS testcase on any device which supports the capability range less than 16.

### Expected Behavior

VTS should maintain & retrieve the enhancement range values from yaml file instead of hardcoding the value ‘16’ in the code.

### Actual Behavior

Enhancement range values got hardcoded in the test case, so the testcase will fail on the device which supports less than 16.

srinivasgtl, 2025-04-06:
This change is already available from HPK 1.4.3 onwards (device setting tag 3.5.0).  Please check and update the issue.

thanushreevani, 2025-04-07:
@anburaj17  Any update with HPK 1.4.3 (device setting tag 3.5.0)

anburaj17, 2025-04-07:
We will verify this in our next release validations and update.

thanushreevani, 2025-04-30:
@anburaj17 any update on this testing

anburaj17, 2025-05-21:
This behavior not observed during the latest release validations (used the version as per 1.4.4 HPK)

**Versions used:**
```
  HALIF version: 4.1.2
  VTS version: 3.5.0
```