rdkcentral/rdk-halif-test-device_settings #392: Bug: Should maintain the separate ‘num_entries’ field for each video codec in yaml file while  executing L2 GetVideoCodecInfo_source testcase

anburaj17, 2025-04-04:
### Problem/Opportunity

Currently the field ‘num_entries’ a common field for all codecs, so each time it’s getting compared the same value with the other codec's driver data & failed.

### Steps to reproduce

Execute 'L2_GetVideoCodecInfo_source' testcase.

### Expected Behavior

VTS should maintain the 'num_entries' field for each codec.

### Actual Behavior

Currently, it's a common field for all codecs.

KarthikeyanR470, 2025-04-23:
@anburaj17 

Actually only one videocodec info is there for realtek device. Hence, the functionality is fine for this scenario. But, the profile will changes for various platforms. As we discussed could you please add a required codecinfo and test it.

KarthikeyanR470, 2025-05-12:
@anburaj17 any update on this testing?

anburaj17, 2025-05-21:
Not observed during the latest release validations. (as per 1.4.4 HPK release)

**Versions used:**
```
 HALIF version: 4.1.2
 VTS version: 3.5.0
```

closed 2025-05-30 by thanushreevani