rdkcentral/rdk-halif-test-device_settings #393: Bug: Received unknown capability values due to uninitialized testcase capability variable.

anburaj17, 2025-04-04:
### Problem/Opportunity

Received some unknown capability values while executing dsGetAudioCapabilities_L1_positive and L2_GetAudioCapabilities testcases. 
Since the variables got uninitialized inside the testcase, it got appended with some garbage value while assigning the capability fields from driver side. So, in RealTek SOC source device it got failed while comparing the expected capability values which present in yaml file.
Same behaviour observed during dsGetEDID testcase too. We’d modified the code as properly initialized, and the issue not seen.

**VTS version used:** 3.1.2

### Steps to reproduce

Execute 'dsGetAudioCapabilities_L1_positive, L2_GetAudioCapabilities and dsGetEDID' testcases

### Expected Behavior

VTS should initialize all the local variables properly before passing it to driver/HAL to retrieve any data

### Actual Behavior

Variables not initialized properly

bhanucbp, 2025-04-07:
@anburaj17 
These are get APIs, and the values are passed by reference (or pointer). It is the driver's responsibility to initialize or update the referenced variable. The driver must not rely on the initial value of the passed-in parameter — whether it contains garbage, zero, or any other data. Providing uninitialized (garbage) values as input is also a valid test case to ensure the driver does not make assumptions about the input content and correctly writes to the provided memory.

anburaj17, 2025-04-15:
@bhanucbp for test coverage purposes, I still believe that checking the behavior with uninitialized(garbage) values is more appropriate under negative test scenarios rather than positive ones. However, in this case, it seems that many uninitialized local variables are being encountered during positive test scenarios. 
I recommend including a separate test step for this as part of the negative test scenarios to ensure clearer test case categorization.

bhanucbp, 2025-04-15:
@anburaj17 
Drivers should not rely on whether it's a positive or negative test case, or whether variables are initialized or uninitialized. These are 'get' functions, and it's the driver's responsibility to return valid output without assuming initialization or appending to the data

srinivasgtl, 2025-04-29:
@anburaj17 , Do you have any further concerns on this?

anburaj17, 2025-05-22:
@srinivasgtl from the above check-in details, it appears that the mentioned local variables have been initialized in the VTS code. However, should we still raise this with the driver's team for further verification? please confirm

kanjoe24, 2025-06-04:
@anburaj17 , can you close this ticket, if you are happy with the changes please.

closed 2025-06-13 by kanjoe24