rdkcentral/rdk-halif-test-device_settings #475: Bug: Test case should expect dsERR_OPERATION_NOT_SUPPORTED when passing junk value if dsSetDisplayframerate is not supported by platform

Abhishek-0412, 2025-07-10:
### Problem/Opportunity

Problem/Opportunity
VTS version : 3.5.0
https://github.com/rdkcentral/rdk-halif-test-device_settings/blob/3.5.0/src/test_l1_dsVideoDevice.c
On dsSetDisplayframerate_L1_negative,  line number 1578 

```
// Step 06: Call dsSetDisplayframerate() with invalid char*
result = dsSetDisplayframerate(handle, "junk");
UT_ASSERT_EQUAL(result, dsERR_INVALID_PARAM);
```

Test step should expect dsERR_OPERATION_NOT_SUPPORTED when passing junk value if dsSetDisplayframerate is not supported by platform 
If the DUT doesn't support the API, then passing any junk value should return dsERR_OPERATION_NOT_SUPPORTED only instead of dsERR_INVALID_PARAM as long as it handles NULL param check.

### Steps to reproduce

Execute the mentioned testcases & observe the output.

### Expected Behavior

Test case should expect dsERR_OPERATION_NOT_SUPPORTED when passing junk value if dsSetDisplayframerate is not supported by platform

### Actual Behavior

Test case expects dsERR_INVALID_PARAM when passing junk value if dsSetDisplayframerate is not supported by platform

KarthikeyanR470, 2025-07-22:
Hi @Abhishek-0412 

As per the device_settings HALIF specification the API **dsSetDisplayframerate**  is supported only in SINK devices and **NOT_SUPPORTED in SOURCE devices**.
**REFER:** https://github.com/rdkcentral/rdk-halif-device_settings/blob/16515f3b8fdbc683164dd58e50a0f7d6ff995d1f/include/dsVideoDevice.h#L388

We have already handled this reported issue in latest VTS release. Please check.
https://github.com/rdkcentral/rdk-halif-test-device_settings/blob/501810747eeff2f1a239f9edbde2300d31a224fc/src/test_l1_dsVideoDevice.c#L1834

Could you please check in latest VTS Release?

Thank you

CC: @srinivasgtl  @kanjoe24

thanushreevani, 2026-10-14:
@Abhishek-0412 , any update on this.