rdkcentral/rdk-halif-test-device_settings #476: Task: L2 dsAudio Update MS12_Capabilities in Source_AudioSettings.yaml

Matthew-Crowson, 2025-07-10:
### Goal

The below L2 dsAudio test is failing in part because of the expectations outlined in the corresponding yaml file.

- test_l2_dsAudio_RetrieveAndVerifyMS12Capabilities

The current _Source_AudioSettings.yaml_ file expects the following capabilities to be reported by the source device:
- `dsMS12SUPPORT_DolbyVolume`
- `dsMS12SUPPORT_DialogueEnhancer`
- `dsMS12SUPPORT_Volumeleveller`
- `dsMS12SUPPORT_DRCMode`

However, the capabilities that the business expects the source device to support are: 
- `dsMS12SUPPORT_DialogueEnhancer`
- `dsMS12SUPPORT_Volumeleveller`

These are the capabilities that will have been verified during the Dolby certification process. Therefore we should update the _Source_AudioSettings.yaml_ file to reflect this expectation.

srinivasgtl, 2025-07-21:
@Matthew-Crowson ,  The config files set on github are just an example as they differ from platform to platform. The respective platform teams should create there own config files with all the supported features.  Hope this helps.

Matthew-Crowson, 2025-07-25:
As per comments, it is not necessary to make this change in the example config files. Therefore I will mark this as resolved.

closed 2025-07-25 by Matthew-Crowson