rdkcentral/rdk-halif-test-device_settings #478: Bug: For VideoCodecInfo configuration is not supported for num_entries more than 1

Abhishek-0412, 2025-07-11:
### Problem/Opportunity

Test Case : L2_GetVideoCodecInfo_source
Test Version : 3.5.0
In line number 319 of  https://github.com/rdkcentral/rdk-halif-test-device_settings/blob/3.5.0/src/test_l2_dsVideoDevice.c#L319
`
UT_ASSERT_EQUAL(info.num_entries, gDSVideoDeviceConfiguration[i].num_codec_entries);
`
num_entries variable must be specific to each codec
For example
num_entries  field for dsVIDEO_CODEC_MPEGHPART2 and num_entries  field for dsVIDEO_CODEC_MPEG4PART10 must be able to be configurable 
but in yaml file (https://github.com/rdkcentral/rdk-halif-test-device_settings/blob/3.5.0/profiles/source/Source_VideoDevice.yaml)
`
VideoCodecInfo:
          num_entries: 1
          VideoCodec0:
            level: 5.1 #HEVC profile level
            #dsVIDEO_CODEC_HEVC_PROFILE_MAIN             = (0x01 << 0)     /*!< 8-bit HEVC video profile. */
            #dsVIDEO_CODEC_HEVC_PROFILE_MAIN10           = (0x01 << 1)     /*!< 10-bit HEVC video profile. */
            #dsVIDEO_CODEC_HEVC_PROFILE_MAINSTILLPICTURE = (0x01 << 2)     /*!< HECV Main Still Picture profile */
            #dsVIDEO_CODEC_HEVC_MAX                      = (0x01 << 3)     /*!< Out of range  */
            profile: 2 #dsVIDEO_CODEC_HEVC_PROFILE_MAIN10
`
num_entries is common for all codecs
it must be separate entry each codecs to validate num_entries  field for each codec

Also in the for loop
`
for(codec = dsVIDEO_CODEC_MPEGHPART2 ; codec < dsVIDEO_CODEC_MAX; )
        {
            if(!(gDSVideoDeviceConfiguration[i].SupportedVideoCodingFormats & codec))
            {
                continue;
            }
            UT_LOG_DEBUG("Invoking dsGetVideoCodecInfo with handle: %ld, codec: %d", handle, codec);
            retStatus = dsGetVideoCodecInfo(handle, codec, &info);
            UT_ASSERT_EQUAL(retStatus, dsERR_NONE);
            UT_LOG_DEBUG("dsGetVideoCodecInfo return status: %d, num_entries: %d", retStatus, info.num_entries);

            UT_ASSERT_EQUAL(info.num_entries, gDSVideoDeviceConfiguration[i].num_codec_entries);
            /* for now hal get info supports only MPEGHPART2 */
            if(codec == dsVIDEO_CODEC_MPEGHPART2)
            {
                UT_LOG_DEBUG("Profile: %d", info.entries->profile);
                UT_ASSERT_EQUAL(info.entries->profile,gDSVideoDeviceConfiguration[i].profile);
                //TODO : kvp profile support for float type is not available now
                //UT_ASSERT_EQUAL(info.entries->level,gDSVideoDeviceConfiguration[i].level);
            }
            codec = (0x01 << (j++));
        } /* for(codec)*/
`
if I give dsVIDEO_CODEC_MPEGHPART2  as unsupported , then it hits the condition
`
 if(!(gDSVideoDeviceConfiguration[i].SupportedVideoCodingFormats & codec))
            {
                continue;
            }
`
but after this codec variable updation is missing, so testcase is stuck in infinite loop
the continue condition must be updated as below so that inifinite loop is avoided
`
for(codec = dsVIDEO_CODEC_MPEGHPART2 ; codec < dsVIDEO_CODEC_MAX; )
        {
            if((gDSVideoDeviceConfiguration[i].SupportedVideoCodingFormats & codec))
            {
                  UT_LOG_DEBUG("Invoking dsGetVideoCodecInfo with handle: %ld, codec: %d", handle, codec);
                  retStatus = dsGetVideoCodecInfo(handle, codec, &info);
                  UT_ASSERT_EQUAL(retStatus, dsERR_NONE);
                  UT_LOG_DEBUG("dsGetVideoCodecInfo return status: %d, num_entries: %d", retStatus, info.num_entries);

                  UT_ASSERT_EQUAL(info.num_entries, gDSVideoDeviceConfiguration[i].num_codec_entries);
                   /* for now hal get info supports only MPEGHPART2 */
                  if(codec == dsVIDEO_CODEC_MPEGHPART2)
                  {
                       UT_LOG_DEBUG("Profile: %d", info.entries->profile);
                       UT_ASSERT_EQUAL(info.entries->profile,gDSVideoDeviceConfiguration[i].profile);
                       //TODO : kvp profile support for float type is not available now
                       //UT_ASSERT_EQUAL(info.entries->level,gDSVideoDeviceConfiguration[i].level);
                   }
             }
             codec = (0x01 << (j++));
        } /* for(codec)*/
`

### Steps to reproduce

Execute the mentioned testcases & observe the output.

### Expected Behavior

num_entries must be separate entry each codecs to validate num_entries  field for each codec

### Actual Behavior

num_entries is common for all codecs

kanjoe24, 2025-07-22:
@Abhishek-0412 , Can you provide more info on your requirements here, may be yaml file that you think should be supported please.

Abhishek-0412, 2025-07-22:
Hi @kanjoe24 

for example if a HAL code is written like this
```
dsError_t dsGetVideoCodecInfo (int handle, dsVideoCodingFormat_t codec, dsVideoCodecInfo_t *info)
{
	if (codec == dsVIDEO_CODEC_MPEGHPART2)
	{
		info->num_entries = 3;
		info->entries[0].profile = dsVIDEO_CODEC_HEVC_PROFILE_MAIN;
		info->entries[0].level = x;
		info->entries[1].profile = dsVIDEO_CODEC_HEVC_PROFILE_MAIN10;
		info->entries[1].level = y;
                info->entries[1].profile = dsVIDEO_CODEC_HEVC_PROFILE_MAINSTILLPICTURE;
		info->entries[1].level = z;
	}
	else
	{
		info->num_entries = 0;
	}
	return dsERR_NONE;
}
```

How to validate in the current framework as num_entries is 3 and for each num_entry different profile is set.

closed 2026-10-23 by kanjoe24