rdkcentral/rdk-halif-test-device_settings #494: Bug: Expect dsHdmiIn L1 APIs to return NOT_SUPPORTED for Source Devices.

Nikitha1602, 2025-08-08:
### Problem/Opportunity

Two test cases are resulting in Parsing error 
- test_l1_dsHdmiIn_positive_dsSetEdidVersion_sink
- test_l1_dsHdmiIn_positive_dsHdmiInSetVRRSupport_sink

The test cases should return OPERATION_NOT_SUPPORTED for the source devices. But the tests attempt to parse configurations (EDID version/VRR ports) for both source and sink devices. Configuration checks should be done for sink devices only.

 

### Expected Behavior

The test cases should return OPERATION_NOT_SUPPORTED without configuration checks

### Actual Behavior

Test cases are failed with Parsing error.

kanjoe24, 2025-08-13:
@Nikitha1602 , this is already taken care.
Please check :
https://github.com/rdkcentral/rdk-halif-test-device_settings/blob/main/src/test_l1_dsHdmiIn.c#L2029
https://github.com/rdkcentral/rdk-halif-test-device_settings/blob/main/src/test_l1_dsHdmiIn.c#L2851
Let me know, if you know more details.

Regards
Jyotsna

kanjoe24, 2025-09-08:
Had a call with @Nikitha1602 , and understood that source yaml doesn't have field : VrrSupportedPorts.
also, need to check for : dsHdmiIn/EdidVersion/

https://github.com/rdkcentral/rdk-halif-test-device_settings/blob/main/profiles/source/Source_HDMIIN.yaml

Error logs:
```
In test_l1_dsHdmiIn_positive_dsSetEdidVersion_sink [01039]

2025-07-18-11:02:39, ERROR , ut_kvp.c,   285 : Parsing Error
2025-07-18-11:02:39, ERROR , ut_kvp.c,   285 : Parsing Error
2025-07-18-11:02:39, LOG   , test_l1_dsHdmiIn.c,  2043 :
Out test_l1_dsHdmiIn_positive_dsSetEdidVersion_sink
 
``` 

```
In test_l1_dsHdmiIn_positive_dsHdmiInSetVRRSupport_sink [01055]

2025-07-18-11:02:39, ERROR , ut_kvp.c,   285 : Parsing Error
2025-07-18-11:02:39, ERROR , ut_kvp.c,   626 : node not found for key = [dsHdmiIn/VrrSupportedPorts] : UT_KVP_STATUS_KEY_NOT_FOUND
2025-07-18-11:02:39, LOG   , test_l1_dsHdmiIn.c,  2905 :
Out test_l1_dsHdmiIn_positive_dsHdmiInSetVRRSupport_sink
 
``` 
The source profile file needs to be updated