rdkcentral/rdk-halif-test-device_settings #504: Bug:  Python test cases are not available for some modules

Anjalimkuriakose, 2025-09-18:
### Problem/Opportunity

Some python testcases are not available for dsAudio, dsHost, dsVideoDevice and dsVideoPort modules.

VTS version used: 3.5.0

### Expected Behavior

Should contain all the L3 python testcases for all modules.

### Actual Behavior

Some of the L3 python testcases are missing for dsAudio, dsHost, dsVideoDevice and dsVideoPort modules.

### Notes (Optional)

Missing testcases 

dsAudio:
1. Enable/Disable Stereo Auto
2. Set ATMOS Output Mode
3. Get ATMOS Capabilities

dsHost:
1. Get SoC ID
2. Get Host EDID

dsVideoDevice:
1. VideoDevice_Term
2. GetHDRCapabilities
3. GetSupportedVideoCodingFormat
4. GetCurrentDisplayframerate
5. GetVideoCodecInfo

dsVideoPort:
1. Get CurrentOutputSettings
2. Get Resolution
3. Get VideoEOTF
4. IsOutputHDR
5. Get HDCPStatus
6. Get HDCPCurrentProtocol
7. Get HdmiPreference
8. Get ColorSpace
9. Get ColorDepth
10. Get HDCPReceiverProtocol
11. Get IgnoreEDIDStatus
12. Get PreferredColorDepth

kanjoe24, 2025-10-03:
@Anjalimkuriakose , Can you cross confirm this on latest release please : https://github.com/rdkcentral/rdk-halif-test-device_settings/tree/6.0.0

Anjalimkuriakose, 2025-10-07:
We have verified with the latest version https://github.com/rdkcentral/rdk-halif-test-device_settings/tree/6.0.0 , but the issue still persists.